f8kgl/swr_power_meter #4: Prototype de coupleur directionnel

f4bjh, 2019-06-29:
Monter une manip pour mesurer les pertes et le facteur de couplage des coupleurs suivants : 

-coupleur en cable coaxial rigide (sur une idée de Pierre F1FDD)
-ligne imprimée sur circuit
-coupleur avec tore
-coupleur du commerce, issu d'une épave de TOS-mètre
-coupleur du commerce issu d'un Wattmètre du monde professionnel (type Bird, ou power 	sensor 4B250 de Thruline

F0EOS, 2019-07-06:
1 calculer la ligne imprimee par qucs 
2 Schema + PCB sous kicad
3 Fabrication (JLPCB ?)

F0EOS, 2019-07-06:
erreur manip

F0EOS, 2019-07-06:
recup parametre circuit FR4 
faire deux CI  deux et trois lignes
http://qucs.sourceforge.net/docs/tutorial/coupler.pdf page 5
(F4BJH voir pour simulation ?)

f4bjh, 2019-07-07:
Pour la simulation sous Qucs, attention à #7

f4bjh, 2020-05-07:
#75

f4bjh, 2020-05-18:
Compte tenu des évolutions prévues en  V0.8, et en particulier d la nouvelle fonctionnalité de xWATT METER prévue, le proto de ligne de couplage ne pourra effectivement se faire qu'avec l'arrivée du 1er proto matériel...

==> fiche repoussée en V1.0

f4bjh, 2022-02-07:
vu avec André, au 600 ohms : cette fiche est mise de côté pour le moment.
il faudra la réouvrir quand nous serons en V1.0

closed 2022-02-07 by f4bjh