f8kgl/swr_power_meter #74: probleme de post

F0EOS, 2019-07-09:
probleme de post  a chaque envoie j'ai ce msg 

[image]

f4bjh, 2019-07-09:
Visiblement, il faut **d'abord** se mettre à jour avant d'uploader une modif quelqueconque

Si je ne dis pas de bêtise, cliquer sur "Git Sync", pour faire la synchronisation

Pour rappel, uplader une modif, se dit aussi "faire un commit", ou faire "git push"
Se synchroniser (menu Git sync dans Windows") se dit aussi "faire un git pull"

f4bjh, 2019-07-09:
Peux -tu vérifer ce que je dis là ? C'est à dire : 

1-faire une modif bidon dans le schéma
2-Faire Git Sync
3-Faire le commit, et vérifier que tu n'as plus de message d'erreurs. Dans le cas contraire, poster içi même la copie d'écran de l'erreur

f4bjh, 2019-07-10:
C'est bon ? C'est résolut du coup ?

f4bjh, 2020-01-05:
On peut fermer ce bug ?

F0EOS, 2020-01-05:
Je suis passer sous W10 donc plus rien sur le pc

J au reinstaller kicad et rien d autre 

Fait que je re installe tout le reste sauf que je me souviens pas, me souviens pas de parametre et mot de passe mais surtout comment ca refonctionne

Donc besoin tuto par mail lol

Donc je ne peux pas te repondre

----- Mail d'origine -----

https://github.com/f8kgl/swr_power_meter/issues/74#issuecomment-570927559

f4bjh, 2020-01-05:
OK.

En attendant que tu remontes ton système, je ferme ce bug. 

Au besoin, on le rouvrira si tu as de nouveaux des problèmes similaires.

closed 2020-01-05 by f4bjh