f8kgl/swr_power_meter #75: Caracteristique FR4

F0EOS, 2019-07-09:
je ne comprends pas la correspondance des caracteristique du FR4 pour paramettre QUBS

De meme en parametre par default j'ai des piste large de 2 ou 3 mm et je trouve cela bizarre

[CARACTERISTIQUE FR4.xlsx](https://github.com/f8kgl/swr_power_meter/files/3374361/CARACTERISTIQUE.FR4.xlsx)

f4bjh, 2019-07-09:
Tout d'abord, pour info : commit 94bb4ea00321671a34b73466ccb6c8338c337daf

**NE PAS UTILISER D'ESPACE DANS LES NOMS DE FICHIERS OU DE REPERTOIRE**
Préferer le caractère "underscore", ou tiret du chiffre 8

f4bjh, 2019-07-09:
Dans QUCS, pourrais-tu me dire quels sont les paramètres qui te manquent exactement ?

F0EOS, 2019-07-09:
tous je ne comprends pas les correspondances entre les caracteristique du FR4 et les parametre de Qucs

f4bjh, 2019-07-09:
Dans QUCS, avec l'outil qui fait la synthèse de la ligne imprimée, si tu passes la souris juste au dessus du nom du paramètre, il te dit de quoi il s'agit.

Exemple : "Er" ----> Permitivité relative ==> Google est ton ami
[https://fr.wikipedia.org/wiki/Permittivit%C3%A9](url)

Et là, tu apprends que la permitivité relative est la même chose que la constante diélectrique.
"Mur" ---> Perméabilité relative ==> Google est ton ami 

[https://fr.wikipedia.org/wiki/Perm%C3%A9abilit%C3%A9_magn%C3%A9tique](url)

et ainsi de suite...

closed 2020-05-18 by f4bjh