f8kgl/swr_power_meter #82: Comment fait-on pour répondre sur Github ?

f4bjh, 2019-07-11:
Quand on poste un issue, GitHub envoie un mail aux personnes concernées, avec le contenu de l'issue. Par ex : 
>
>
> §3.1.1
> Dimensions = 155x80x100 (mm
>
> Est-ce suffisant ?
>
> —
> You are receiving this because you were assigned.
> Reply to this email directly, view it on GitHub, or mute the thread.

Pour y répondre, il faut : 
1-Cliquer sur "view it on GitHub"
2-Répondre dans l'espace commentaire, située en dessous 
[image]

f4bjh, 2019-07-11:
Est-ce que ça répond à ta question ?

F0EOS, 2019-07-11:
pas connecter donc pas de fenêtre réponse

closed 2019-07-11 by f4bjh