f8kgl/swr_power_meter #94: EEschéma remonte des erreurs lors du test des règles électriques

f4bjh, 2020-01-08:
Rapport d'erreur (mer. 08 janv. 2020 01:22:08 CET, Encodage UTF8 )

***** Feuille /
ErrType(3): Pin connectée à d'autres pins, mais aucune pin pour la piloter
    @(207,01 mm, 133,35 mm): Pin 1 (Power input) du composant #PWR03 non pilotée (Net 1).
ErrType(3): Pin connectée à d'autres pins, mais aucune pin pour la piloter
    @(196,85 mm, 116,84 mm): Pin 1 (Entrée) du composant U1 non pilotée (Net 6).
ErrType(3): Pin connectée à d'autres pins, mais aucune pin pour la piloter
    @(80,01 mm, 29,21 mm): Pin 1 (Power input) du composant U2 non pilotée (Net 10).
ErrType(2): Pin non connectée (utiliser un marqueur de "non connexion" pour supprimer cette erreur)
    @(144,78 mm, 57,15 mm): La pin 1 (Entrée) du composant U4 est non connectée.
ErrType(2): Pin non connectée (utiliser un marqueur de "non connexion" pour supprimer cette erreur)
    @(168,91 mm, 57,15 mm): La pin 10 (Entrée) du composant U4 est non connectée.
ErrType(2): Pin non connectée (utiliser un marqueur de "non connexion" pour supprimer cette erreur)
    @(144,78 mm, 64,77 mm): La pin 4 (Entrée) du composant U4 est non connectée.
ErrType(2): Pin non connectée (utiliser un marqueur de "non connexion" pour supprimer cette erreur)
    @(144,78 mm, 67,31 mm): La pin 5 (Entrée) du composant U4 est non connectée.
ErrType(2): Pin non connectée (utiliser un marqueur de "non connexion" pour supprimer cette erreur)
    @(168,91 mm, 59,69 mm): La pin 9 (Entrée) du composant U4 est non connectée.
ErrType(2): Pin non connectée (utiliser un marqueur de "non connexion" pour supprimer cette erreur)
    @(201,93 mm, 91,44 mm): La pin 18 (Entrée) du composant U1 est non connectée.
ErrType(2): Pin non connectée (utiliser un marqueur de "non connexion" pour supprimer cette erreur)
    @(219,71 mm, 116,84 mm): La pin 10 (Bidirectionnel) du composant U1 est non connectée.
ErrType(2): Pin non connectée (utiliser un marqueur de "non connexion" pour supprimer cette erreur)
    @(204,47 mm, 91,44 mm): La pin 17 (Entrée) du composant U1 est non connectée.
ErrType(2): Pin non connectée (utiliser un marqueur de "non connexion" pour supprimer cette erreur)
    @(199,39 mm, 91,44 mm): La pin 19 (Entrée) du composant U1 est non connectée.
ErrType(2): Pin non connectée (utiliser un marqueur de "non connexion" pour supprimer cette erreur)
    @(199,39 mm, 116,84 mm): La pin 2 (Entrée) du composant U1 est non connectée.
ErrType(2): Pin non connectée (utiliser un marqueur de "non connexion" pour supprimer cette erreur)
    @(201,93 mm, 116,84 mm): La pin 3 (Bidirectionnel) du composant U1 est non connectée.
ErrType(2): Pin non connectée (utiliser un marqueur de "non connexion" pour supprimer cette erreur)
    @(204,47 mm, 116,84 mm): La pin 4 (Bidirectionnel) du composant U1 est non connectée.
ErrType(2): Pin non connectée (utiliser un marqueur de "non connexion" pour supprimer cette erreur)
    @(214,63 mm, 116,84 mm): La pin 8 (Bidirectionnel) du composant U1 est non connectée.
ErrType(2): Pin non connectée (utiliser un marqueur de "non connexion" pour supprimer cette erreur)
    @(217,17 mm, 116,84 mm): La pin 9 (Bidirectionnel) du composant U1 est non connectée.

** messages ERC: 17  Erreurs 0  Avertissements 17

f4bjh, 2020-01-08:
Pour la plupart, je pense qu'il suffit de placer un marqueur de non connexion...
Mais je n'ai pas tout regardé dans le détail

f4bjh, 2020-03-09:
https://github.com/f8kgl/swr_power_meter/commit/d068d41402ae436b3c028524527057c7facadf2c

f4bjh, 2020-03-09:
Il y a du mieux, il y a moins d'erreur, mais il en reste : 

Rapport d'erreur (lun. 09 mars 2020 20:52:09 CET, Encodage UTF8 )

***** Feuille /
ErrType(3): Pin connectée à d'autres pins, mais aucune pin pour la piloter
    @(140,39 mm, 65,25 mm): Pin 1 (Power input) du composant #PWR0101 non pilotée (Net 2).
ErrType(3): Pin connectée à d'autres pins, mais aucune pin pour la piloter
    @(196,85 mm, 116,84 mm): Pin 1 (Entrée) du composant U1 non pilotée (Net 3).
ErrType(3): Pin connectée à d'autres pins, mais aucune pin pour la piloter
    @(80,01 mm, 29,21 mm): Pin 1 (Power input) du composant U2 non pilotée (Net 8).
ErrType(2): Pin non connectée (utiliser un marqueur de "non connexion" pour supprimer cette erreur)
    @(204,47 mm, 91,44 mm): La pin 17 (Entrée) du composant U1 est non connectée.
ErrType(2): Pin non connectée (utiliser un marqueur de "non connexion" pour supprimer cette erreur)
    @(201,93 mm, 91,44 mm): La pin 18 (Entrée) du composant U1 est non connectée.
ErrType(2): Pin non connectée (utiliser un marqueur de "non connexion" pour supprimer cette erreur)
    @(199,39 mm, 91,44 mm): La pin 19 (Entrée) du composant U1 est non connectée.
ErrType(2): Pin non connectée (utiliser un marqueur de "non connexion" pour supprimer cette erreur)
    @(199,39 mm, 116,84 mm): La pin 2 (Entrée) du composant U1 est non connectée.
ErrType(2): Pin non connectée (utiliser un marqueur de "non connexion" pour supprimer cette erreur)
    @(201,93 mm, 116,84 mm): La pin 3 (Bidirectionnel) du composant U1 est non connectée.
ErrType(2): Pin non connectée (utiliser un marqueur de "non connexion" pour supprimer cette erreur)
    @(204,47 mm, 116,84 mm): La pin 4 (Bidirectionnel) du composant U1 est non connectée.
ErrType(2): Pin non connectée (utiliser un marqueur de "non connexion" pour supprimer cette erreur)
    @(214,63 mm, 116,84 mm): La pin 8 (Bidirectionnel) du composant U1 est non connectée.
ErrType(2): Pin non connectée (utiliser un marqueur de "non connexion" pour supprimer cette erreur)
    @(217,17 mm, 116,84 mm): La pin 9 (Bidirectionnel) du composant U1 est non connectée.
ErrType(2): Pin non connectée (utiliser un marqueur de "non connexion" pour supprimer cette erreur)
    @(144,78 mm, 57,15 mm): La pin 1 (Entrée) du composant U4 est non connectée.
ErrType(2): Pin non connectée (utiliser un marqueur de "non connexion" pour supprimer cette erreur)
    @(168,91 mm, 57,15 mm): La pin 10 (Entrée) du composant U4 est non connectée.
ErrType(2): Pin non connectée (utiliser un marqueur de "non connexion" pour supprimer cette erreur)
    @(219,71 mm, 116,84 mm): La pin 10 (Bidirectionnel) du composant U1 est non connectée.
ErrType(2): Pin non connectée (utiliser un marqueur de "non connexion" pour supprimer cette erreur)
    @(144,78 mm, 64,77 mm): La pin 4 (Entrée) du composant U4 est non connectée.
ErrType(2): Pin non connectée (utiliser un marqueur de "non connexion" pour supprimer cette erreur)
    @(144,78 mm, 67,31 mm): La pin 5 (Entrée) du composant U4 est non connectée.
ErrType(2): Pin non connectée (utiliser un marqueur de "non connexion" pour supprimer cette erreur)
    @(168,91 mm, 59,69 mm): La pin 9 (Entrée) du composant U4 est non connectée.

** messages ERC: 17  Erreurs 0  Avertissements 17

F0EOS, 2020-03-09:
Je vais voir avec kicad comment on fait pour le signale en non connecté  

J ai pas vu l'option je vais la chercher

----- Mail d'origine -----

https://github.com/f8kgl/swr_power_meter/issues/94#issuecomment-596748064

f4bjh, 2020-04-23:
MAJ du DRC suite au commit https://github.com/f8kgl/swr_power_meter/commit/9f2989986ff7ac75562613929bf327e1f65362b7

***** Feuille /
ErrType(3): Pin connectée à d'autres pins, mais aucune pin pour la piloter
    @(72,39 mm, 29,21 mm): Pin 1 (Power input) du composant U2 non pilotée (Net 3).
ErrType(3): Pin connectée à d'autres pins, mais aucune pin pour la piloter
    @(196,85 mm, 116,84 mm): Pin 1 (Entrée) du composant U1 non pilotée (Net 5).
ErrType(3): Pin connectée à d'autres pins, mais aucune pin pour la piloter
    @(165,10 mm, 82,55 mm): Pin 1 (Power input) du composant #PWR0106 non pilotée (Net 7).

F0EOS, 2020-04-23:
sALUT FABRICE 

comprends pas trop les erreurs 
la pin 1 de U1 est bien connecter a la pin 1 de U3 alors que ton message l'indicque en erreur

en fait je ne sais pas a quoi cela correspond

Andre
Bien recuperer ta livraison 

----- Mail original -----

MAJ du DRC suite au commit 9f29899

f4bjh, 2020-04-27:
avec le passage en V0.7, les DRC précédents n'ont plus de sens

closed 2020-04-27 by f4bjh