f8kgl/swr_power_meter #98: connection MAX11100

F0EOS, 2020-04-04:
est ce que la borne 6 ( ref )  du max11100  il y a que le C4 ou faut il aussi autre chose a connecter dessus

Voir ligne sur le schéma dans le vide, je ne sais pas si c'est une erreur de ma part ou si j'ai oublier de connecter qque chose a la fin

f4bjh, 2020-04-04:
OK, je vais regarder ça très prochainement.

F0EOS, 2020-04-04:
j'ai cemessage quand j'ouvre le projet avec kicad

----- Mail original -----

f4bjh, 2020-04-05:
Il faut créer une autre fiche pour cela...sinon, on risque de tout mélanger...

f4bjh, 2020-04-23:
Il y a des erreurs de cablage sur U4 (MAX11100)

1/il faut relier SCLK(pin 10), CS (pin 9), OUT (pin 1) à des pins encore libre du PIC.
Parmi les disponibles il y a : 2,3,4, 8,9,10,17,18,19
Choisir celle qui facilite le routage.

2/relier AGND (pin 8) à la masse
3/déconnecter la pin 1 du PIC pour la mettre sur la pin AIN(pin 5) du MAX11100

f4bjh, 2020-04-23:
Attention : Erreur sur le point 3/
Laisser la pin 1 du PIC telle quelle est cablée aujourd'hui
Mais cabler la pin 5 de U3 sur la pin 5 de U4

F0EOS, 2020-04-23:
a l'occas on s'apelle  parce que sur U4 il y a que 10 bornes

pas de borne 17-18-19

----- Mail original -----

f4bjh, 2020-04-23:
les pin 2,3,4, 8,9,10,17,18,19 sont les pin disponibles du PIC
il faut relier les pin 1, 9 10 de U4 à l'une des pin du PIC encore disponibles. Choisir celles qui facilitent le routage

f4bjh, 2020-04-27:
En prévision du tag V0.6, cette fiche ne sera pas résolue (évolution de U4 à définir)

closed 2020-04-27 by f4bjh