f8kgl/swr_power_meter #101: Kicad fait une erreur au démarrage

f4bjh, 2020-04-23:
[image]

f4bjh, 2020-04-23:
Visiblement, est apparu avec ce commit : https://github.com/f8kgl/swr_power_meter/commit/f18413c56c2a7412bc5f6d7ad0c81c7bb11e8d8b#diff-fa0e92bc9e44e93ac15648b0c116433a

On dirait que le résultat d'un diff a été livré dans ce fichier.
=> suppression manuelle

f4bjh, 2020-04-26:
André, tu me confirmes que ce message d'erreur a bien disparu chez toi aussi ?

f4bjh, 2020-04-27:
André, est-ce que tu as toujours cette erreur ? (il faut que tu viennes voir cette fiche sur Github, pour me le confirmer ou pas...j'ai besoin de savoir si je peux ferme cette fiche ou pas, pour poser un tag)

F0EOS, 2020-04-27:
je n'ai plus ce message d'erreur mais un autre

closed 2020-04-27 by F0EOS