f8kgl/swr_power_meter #106: Impossible de livrer sur github avec Turtoise git

f4bjh, 2020-04-26:
A de nombreuses tentatives de livraison avec Turtoise git, cette erreur apparaît : 

error: failed to push some refs to 'https://github.com/f8kgl/swr_power_meter'
hint: Updates were rejected because the remote contains work that you do
hint: not have locally. This is usually caused by another repository pushing
hint: to the same ref. You may want to first integrate the remote changes
hint: (e.g., 'git pull ...') before pushing again.
hint: See the 'Note about fast-forwards' in 'git push --help' for details.

Mais ce n'est pas systématique

f4bjh, 2020-04-26:
J'ai ajouté un texte en plein milieu du schéma.
A voir si tu arrives à bien le télécharger avec Turtoise Git.
Le message contient une phrase du style "Texte de test pour André. A supprimer quand la fiche #106 sera résolue"
Ensuite, le supprimer, et relivrer pour voir si ça marche.

f4bjh, 2020-04-26:
référence https://github.com/f8kgl/swr_power_meter/commit/920d66d8e337184d50d51a6e84954f5b0fb8cc69 ci-dessus erronée

F0EOS, 2020-04-26:
nickel pas eu de souci cette fois  

Pourvu que ca dure ....

closed 2020-04-26 by F0EOS