f8kgl/swr_power_meter #120: Driver ADC

f4bjh, 2020-04-28:


f4bjh, 2020-04-28:
commencer par implémenter l'ADC sous gpsim

f4bjh, 2020-04-28:
Mon analyse pour la mise en oeuvre du LTC2305 sous gpsim : 

Exemple : composant i2c2par

Côté I2C, les données doivent passer, à un moment donné, par gpsim-0.30.0/src/i2c-ee.cc qui fait appel à get_data ou put_data.
Ces fonctions sont codées dans modules/i2c2par.cc, qui va charcher les données sur un "IOPort"

Du coup, ce qui vaudrait le coup ce serait : 
1/créer une branche sur le projet pour les developpements i2c
2/dans cette branche : 
2.1/mettre en oeuvre un driver i2c2par dans le PIC, qui récupère les valeurs du IOPort (fixé à une valeur sur 8 bits, par 01010101 (ou autre peu importe) => vérifier que l'on arrive bien à récupérer la valeur attendue
2.2/créer un nouveau composant stricitement identique à i2c2par, et refaire cette manip
2.3/faire évoluer le composant pour changer IOPort en un port qui reçoit une tension et la converti directement en hexa. La récupérer dans le PIC.

Question : faut-il refaire un make et make install de gpsim après modif d'un composant ?
j'ai installé gpsim par apt-get. Faut-il le désinstaller du coup ?

f4bjh, 2020-04-30:
Régression avec ce commit https://github.com/f8kgl/swr_power_meter/commit/15b4c0a44294a39fe523f65b056f44dc60ba33b1

[image]

Retour du pb de PCLATH ? #122

f4bjh, 2020-05-02:
-->1/créer une branche sur le projet pour les developpements i2c
implémenté et validé (branche i2c_test)

-->2/dans cette branche :
-->2.1/mettre en oeuvre un driver i2c2par dans le PIC, qui récupère les valeurs du IOPort (fixé à une valeur sur 8 bits, par 01010101 (ou autre peu importe) => vérifier que l'on arrive bien à récupérer la valeur attendue

en cours
pb n°1 : impossible de driver SDA et SCL de i2c2par #131 
pb n°2 : plantage du PIC. Bloqué en PCL=0x0CE0 #130 

-->2.2/créer un nouveau composant stricitement identique à i2c2par, et refaire cette manip
-->2.3/faire évoluer le composant pour changer IOPort en un port qui reçoit une tension et la converti directement en hexa. La récupérer dans le PIC.

Non commencés

f4bjh, 2020-05-04:
-->2/dans cette branche :
-->2.1/mettre en oeuvre un driver i2c2par dans le PIC, qui récupère les valeurs du IOPort (fixé à une valeur sur 8 bits, par 01010101 (ou autre peu importe) => vérifier que l'on arrive bien à récupérer la valeur attendue

pb n°3 : la sortie de i2c2par vaut 0xD5 au lieu de 0xAA #132

f4bjh, 2020-05-04:
pb n°4 : SCL=400kHz ??? #133

f4bjh, 2020-05-04:
Pour créer une nouvelle librairie et un nouveau composant : 
1/créer un ficher source librairie.cc (sur le modèle de libgpsim_modules.cc)
2/créer un fichier source module.cc (sur le modèle de i2c2par.cc)
3/Adapter le Makefile.am (ajout de la librairie et du module)
4/
$cd .../gpsim.0.31
$ autoreconf
$./configure
$make
$sudo make install
$sudo /sbin/ldconfig

f4bjh, 2020-05-04:
-->1/créer une branche sur le projet pour les developpements i2c
implémenté et validé (branche i2c_test)

-->2/dans cette branche :
-->2.1/mettre en oeuvre un driver i2c2par dans le PIC, qui récupère les valeurs du IOPort (fixé à une valeur sur 8 bits, par 01010101 (ou autre peu importe) => vérifier que l'on arrive bien à récupérer la valeur attendue

implémenté et validé (branche i2c_test)

-->2.2/créer un nouveau composant stricitement identique à i2c2par, et refaire cette manip
implémenté et validé (branche i2c_test)
avec malgré un petit souci non rédhibitoire a priori : #136 
-->2.3/faire évoluer le composant pour changer IOPort en un port qui reçoit une tension et la converti directement en hexa. La récupérer dans le PIC.

f4bjh, 2020-05-05:
Pour rappel içi, pour placer un scope sous gpsim : 

**gpsim> scope.ch0="porta3"
**gpsim> scope.ch1="porta2"

f4bjh, 2020-05-05:
reste à implémenter : 

1/CH1
2/les octets de commande du ltc2305

F0EOS, 2020-05-05:
c est quoi tout ces composant que je vois defile

Ca sera a rajouter ?

Il va falloir faire du 2 face on ne pourrapas faire tout tenir sur une

----- Mail d'origine -----

https://github.com/f8kgl/swr_power_meter/issues/120#issuecomment-624251174

f4bjh, 2020-05-06:
> c est quoi tout ces composant que je vois defile Ca sera a rajouter ? Il va falloir faire du 2 face on ne pourrapas faire tout tenir sur une
> […](#)
> ----- Mail d'origine ----- De: Fabrice DECROP [...]

Tu n'es pas concerné par tout ça...c'est la simulation sous gpsim...

f4bjh, 2020-05-06:
il manque encore la table de calibration de test

f4bjh, 2020-05-11:
référence à 142 ci-dessus eronnée

f4bjh, 2020-05-11:
https://github.com/f8kgl/swr_power_meter/commit/5d00e68406ee17e5fcb0c368d43a4d25a3c3e9fb

restrictions non bloquantes d'ores et déjà connues (liées à gpsim a priori) : 
-pas de tempo pour la conversion (normalement, 1,3us) #180
-1ère trame renvoie NACK #134 #159

A faire apparaître dans la doc d'implémentation V0.7

f4bjh, 2020-05-13:
et à reporter sur la doc V0.8

closed 2020-05-26 by f4bjh