f8kgl/swr_power_meter #123: mise a jour DATASHEET

F0EOS, 2020-04-30:
recuperation datasheet BAT54 et AD5175

Manque LT818 je compte sur toi pour le mettre  dans le dossier

je le trouve pas, il y a des LT 8XX mais pas de 818

f4bjh, 2020-05-01:
>recuperation datasheet BAT54 et AD5175
Attention : éviter de mettre des espaces dans les noms de fichier.
De plus, la datasheet que tu as récupéré de la BAT54, n'est pas celle que j'ai mise dans la documentation d'implémentation : https://github.com/f8kgl/swr_power_meter/blob/master/doc/SWR_POWER_METER_F8KGL_implementation_v0.7.pdf

Je corrige ça dans le commit à suivre

f4bjh, 2020-05-01:
--->Manque LT818 je compte sur toi pour le mettre dans le dossier

je le trouve pas, il y a des LT 8XX mais pas de 818

Elle se trouve içi : https://github.com/f8kgl/swr_power_meter/blob/master/src/hw/doc/lt1818.pdf

f4bjh, 2020-05-01:
Petite remarque anodine (parce que je pense que tu ne l'avais peut être pas compris a priori) : 

Les liens que j'ai mis dans les commentaires ci-dessus, sont des liens vers le serveur github.
MAIS
ceux sont aussi des liens qui pointent sur les répertoires du projet.

C'est à dire que : 
https://github.com/f8kgl/swr_power_meter/blob/master/doc/SWR_POWER_METER_F8KGL_implementation_v0.7.pdf
tu le trouveras chez toi dans swr_power_meter/doc/SWR_POWER_METER_F8KGL_implementation_v0.7.pdf

et 
https://github.com/f8kgl/swr_power_meter/blob/master/src/hw/doc/lt1818.pdf
tu le trouveras chez toi dans swr_power_meter/src/hw/doc/lt1818.pdf

Tu comprends ?

closed 2020-05-02 by f4bjh