f8kgl/swr_power_meter #126: Faut-il créer une branche de test pour les développements HW ?

f4bjh, 2020-05-01:
Git permet la création de branche : 

En partant d'une version à un instant donné, on peut créer un dépot de test supplémentaire, dans lequel on peut faire des livraisons de tests qui n'affectent pas le dépot principal.

Lorsque les livraisons de tests sont terminées et validées, on peut fusionner le dépot de test avec le dépot principal.
Ce dépot supplémentaire serait dédié aux livraisons kicad par exemple.

Est-ce que ça t'intéresse André ?

f4bjh, 2020-05-06:
est-ce que tu veux une branche spéciale pour toi pour faire des livraisons à part ?

F0EOS, 2020-05-06:
non c'etait juste pour les video plus facile a retrouver mais tu as dis pas possible donc non je reste comme ca pour le moment 

j'ai plein de sauvegarde sur le pc pour les test au cas ou

closed 2020-05-06 by f4bjh