f8kgl/swr_power_meter #168: Ajouter le LT1818 au schéma

f4bjh, 2020-05-10:
Poser le composant LT1818 au schéma (sans autre ajout supplémentaire pour le moment)

Même principe que la #164

F0EOS, 2020-05-10:
pas dispo dans la bibliothèque de KICAD

faut que je le face 

tu le veux en MSOP 8 ou T-SOT23 ?

f4bjh, 2020-05-10:
exact...la doc n'est pas assez précise...

je complète cela dès que possible. #171 bloquant

f4bjh, 2020-05-11:
#172 bloquant

f4bjh, 2020-05-11:
référence ci-dessus erronée

f4bjh, 2020-05-13:
Ahhh y est !!!

On en était là, tu te souviens ?

Tu trouveras tout ce qu'il te manque (a priori, et sauf oubli involontaire de ma part) içi : 

https://github.com/f8kgl/swr_power_meter/blob/master/doc/SWR_POWER_METER_F8KGL_specifications_V0.7.pdf

Pour le moment, dans cette fiche, il n'est question "QUE" de poser le composant LT1818 au schéma (sans autre ajout supplémentaire pour le moment)

f4bjh, 2020-05-13:
#118

f4bjh, 2020-05-16:
Very very good job !!! :-)

closed 2020-05-18 by f4bjh