f8kgl/swr_power_meter #191: LIBRAIRIE PROJET

F0EOS, 2020-05-13:
lors de la création du LT818 et sont enregistrement j'avais deux dossier librairie projet PK ??

dans les post  librairie projet n'a pas ete poste, il ne sais pas sélectionner automatiquement normal ??

f4bjh, 2020-05-13:
Sous Kicad, dans l'éditeur de schéma, peux tu aller dans "Préférences->Configurer les librairies de symbole ->Librairies Spécifiques au Projet" ? 

Et faire une copie d'écran

f4bjh, 2020-05-13:
#168

f4bjh, 2020-05-14:
André : 

**1/Problème de librairie**
Sous Kicad, dans l'éditeur de schéma, peux tu aller dans "Préférences->Configurer les librairies de symbole ->Librairies Spécifiques au Projet" ?

Et faire une copie d'écran

**2/Erreur nom de composant**
attention, le nom du composant ajouté est erroné.
Il faut mettre LT1818 et non pas LT818

**3/Oubli de la chaine "REF**
Si tu regardes bien les specs, il faut deux fois le composant AD5175, et deux fois le composant LT1818

Voilà, voilà

f4bjh, 2020-05-16:
Je suppose donc que ton problème de librairie est finalement résolu.

closed 2020-05-18 by f4bjh