f8kgl/swr_power_meter #202: message erreur police

F0EOS, 2020-05-17:
une erreur de police a l'ouverture de SWR_POWER_METER_F8KGL_specifications 

cela s'affiche quand j'arrive a la page 21

[image]

F0EOS, 2020-05-17:
j'ai oublier j'ai plein de tbd ou Tbd rouge sur le pdf en question 

Ça correspond a quoi ?

f4bjh, 2020-05-18:
en fait "tbd" ça veut dire (en anglais) "To Be Defined"...
Par exemple, en page 19 tu trouves : 

Le bloc « Alimentation » est composé de tbd modules :
tbd

c'est à dire qu'à l'heure actuelle, je ne sais pas encore exactement combien il y aura de composant dans le bloc "Alimentation" du schéma...bon, cet exemple n'est pas encore super parlant...parce qu'en fait on sait déjà qu'il y a un régulateur 5V...et horsmis cela, je vois pas trop ce qu'on pourrait enlever ou rajouter...faudrait que je fasse un petit effort pour enlever tous les "tbd" et entièrement tout spécifier.
A vrai dire ce n'est pas ma priorité à l'heure actuelle. Je suis surtout sur les algorithmes de calibration, qui sont des sujet beaucoup plus complexes pour moi...

Autrement dit, et en résumé, tout partout ou tu vois des "tbd" en rouge, ça veut simplement dire que je n'ai pas encore assez travaillé ce sujet là en particulier. Mais qu'un jour ou l'autre, ce sujet là sera précisé et entièrement spécifié.

Ca veut dire qu'il y aura encore des modifs de schéma ? C'est possible oui...

f4bjh, 2020-05-18:
S'agissant du problème d'extraction de police, il semblerait que cela vient d'un paramètre de configuration à placer dans Libreoffice, qui me sert d'éditeur de document.

C'est un bug. Je le corrigerai (du moins j'essaierai...) quand j'aurais traiter tous les autres bugs (et j'en ai quelque uns ;-)

f4bjh, 2020-05-27:
Malheureusement, je crains qu'il ne soit pas possible de résoudre ce problème : 

https://ask.libreoffice.org/en/question/190046/exported-pdf-gives-liberationserif-font-error-in-adobe-acrobat-reader/

closed 2020-06-09 by f4bjh