f8kgl/swr_power_meter #318: Petite erreur au schéma

f4bjh, 2021-02-17:
Voici une fiche pour commencer avec github, Kicad, etc...
Il y a une petite bourde au schéma : il y a un fil en l'air sur le fil de donnée du bus I2C

[image]
[image]

Il faut connecter ces 2 petits fils en l'air :-)

bloquant pour #289

MonFrancoue, 2021-02-20:
le fil est reconnecte 
j espere maintenant que le fichier est mis a jour

f4bjh, 2021-02-22:
aïe...je te confirme ce que l'on se disait avec André : à ce jour, pas vu, ni reçu de mise à jour du schéma pour cette fiche :-p

MonFrancoue, 2021-02-22:
Bonsoir Fabrice

Il semble que ce soit une histoire de droit insuffisant pour livrer les
modifications

f4bjh, 2021-02-23:
C'est bon, ça passe. Je passe cette fiche en "terminé"

closed 2021-03-08 by f4bjh