f8kgl/swr_power_meter #356: Revoir le placement du détecteur HF

f4bjh, 2021-03-05:
Il y a un (assez gros) travail à faire autour du placement du détecteur HF et du LT1818

Je conseillerai de faire de la manière suivante : 

1/déplacer le détecteur HF 
[image]

2/Idem sur la voie REF
3/Rotation de 180° des détecteurs
4/rotation des LT1818 pour avoir des connections en directe

5/améliorer le +d'alim du pont diviseur : dans l'idée d'avoir une connexion plus courte, et surtout (surtout !!!) éviter les boucles d'alim au routage
[image]

La philosophie générale que je préconise est la suivante : 
faire un schéma qui se rapproche le plus possible du PCB (d'où certains fils du schéma qui passe EN DESSOUS des composants). C'est effectivement de cette façon que le routage doit se faire, pour que les fils ne se croisent pas.
Du coup, il faut faire le placement des composants COMME SI on était déjà en train de router le PCB pour de vrai.

f4bjh, 2021-03-05:
#289

f4bjh, 2021-03-06:
Ah ouais !!! C'est niquel pour U5.

Reste à faire la même chose sur U3

MonFrancoue, 2021-03-06:
[image: image.png]  si on laisse le composant comme cela le logiciel est il
capable de le placer correctement ou bien vas t il croire que le composant
est monté à l'envers ?

f4bjh, 2021-03-06:
Je ne connais pas assez Kicad pour evaluer la qualité du placement/routeur automatique.

En tout état de cause, ce qui compte, quelle que soit la valeur de Kicad, ceux sont les connexions électriques, qui elles doivent être correctes au schéma. A mons d'un gros bug chez Kicad, si la pin n°1 de U1 est connecté à une fiche SMA (c'est un exemple), on a l'assurance que cera le cas dans le mesh sur le PCB. La notion de composant à l'envers n'a, a priori, pas trop de sens à ce niveau.

Par contre, je conseille toujours de ne pas se fier aux placement/routeur automatique. Car le placement/routage est un sujet EXTREMENT délicat. S'ils ne sont pas de qualité, des problèmes fonctionnels important peuvent se présenter au final. Le pacement/routage est affaire de compromis. Il y a évidemment quelques règles incontournables : des connexions courtes, pas de boucles, éviter les couplages entre siganux sensibles et bus numériques haute vitesse, alim, etc...
Mais on ne fait pas toujours tout ce qu'on veut. C'est pour ça qu'il faut parfais faire des compromis. Et c'est là que les alogorithmes automatiques des placements/routage montrent bien des faiblesses.

Bref...tout ça pour dire que (et pour répondre à ta question) : normalement, non, il n'y a pas de raison a priori pour qu'il place le composant à l'envers. Par contre, je déconseille vivement d'utiliser les placeurs/routeurs automatiques. C'est la raison pour laquelle je vous demande de faire un schéma qui se rapproche déjà du PCB. Car, alors, il suffira de placer les composants au même endroits géométriques que le schéma. De tracer les pistes du PCB de la même manière que le schéma. S'il n'y a pas de piste qui se croisent au schéma, il n'y aura pas non plus de piste qui se croisent au moment du routage. Autre avantage de cette méthode (qui est la méthode F1FDD), c'est qu'en faisant le placement déjà au schéma, c'est que c'est plus simple au schéma (que sur le PCB en live). En effet, au schéma, c'est plus lisible, et on a une vision plus fonctionnelle que sur le PCB en live. Moins de risque de se plante a priori.

MonFrancoue, 2021-03-07:
Merci Fabrice

closed 2021-03-08 by f4bjh