f8kgl/swr_power_meter #358: Pb de connecteurs

f4bjh, 2021-03-05:
Il y a un problème ave les connecteurs, et je n'arrive pas à trouver des connecteurs qui vraiment pourraient faire l'affaire.

Je liste içi les contraintes : 

1/Connecteurs CMS
Aujourd'hui, au schéma, on a des connecteurs traversant. C'est un tout petit eu embêtant, car ça veut dire qu'on est obligé de détourer le plan de masse sur le PCB. Ca a surtout l'inconvénient d'augmenter le QSJ
Il nous faudrait des connecteurs CMS

2/Connecteurs avec détrompeur
pour empêcher que les OM branche à l'envers. Tout simplement

3/Connecteur pour lesquels le (ou les) câbles ne se débranchent pas en bougeant le tosmètre...
Hormis peut-être pour le connecteur de programmation.

4/Des connecteurs pour lesquels on sait "facilement" fabriquer des câbles et des nappes
En effet, il faut pouvoir fabriquer "facilement" les cables LCD, programmateur, pile, alim, etc...
Connecteur à sertir pour les câbles ? Comment ça marche ? Ca se fait facilement ?
Connecteurs à souder pour les câbles ? 

Il faut 2 connecteurs 2 points (pour l'inverseur du pack pile, et pour le bouton poussoir)
Il faut 1 connecteur 3 points (pour l'alim 12V)
Il faut un connecteur 6 points pour le connecteur de programmation
Il faut un connecteur 8 points pour le LCD

Pour les piles, il y a avait ça : https://fr.rs-online.com/web/p/embases-circuits-imprimes/4838461/ et https://fr.rs-online.com/web/p/boitiers-de-connecteurs-pour-circuits-imprimes/6795363/
Mais c'est traversant côté PCB, et à sertir côté câble ==> bof bof

Pour le LCD, il y aurait éventuellement ça : https://fr.rs-online.com/web/p/embases-circuits-imprimes/8633295/
Mais que faire côté nappe LCD ?

Pour la programmation, il y a ça : https://fr.rs-online.com/web/p/embases-circuits-imprimes/7676695/
Mais idem, comment faire le câble de programmation ?

f4bjh, 2021-03-05:
#306

f4bjh, 2021-03-05:
#289

F0EOS, 2021-03-05:
j'ai trouve les connecteur suivant qui pourrais servir

connecteur HARVIN
https://fr.rs-online.com/web/p/boitiers-de-connecteurs-pour-circuits-imprimes/1707061/
ou ce type la 
https://fr.rs-online.com/web/p/boitiers-de-connecteurs-pour-circuits-imprimes/6812824/

bien sur avec le nombre de bornes adequat

f4bjh, 2021-04-01:
il va y avoir un peu le même pb que pour les fiches SMA : les connecteurs LCD et programmation sont traversants => il faudrait détourer le plan de masse. Chose à éviter au maximum !!!

Il faut trouver des connecteurs CMS pour brancher le LCD et le connecteur de programmation.

f4bjh, 2022-01-12:
@F0EOS tu peux regarder cette fiche : #358 ?

et me dire si d'après toi, elle est résolue ?

closed 2022-04-30 by f4bjh